relativty/Relativty #4: Massively Extending game/software support

LifeIsStrange, 2018-01-21:
Well, first of all thank you for this open source project, you seems to be a funny team and I hope this will become famous :)

But it would be really Nice to add support for openXR (the next crossplatform, crosshardware native standard) 
https://www.khronos.org/openxr

For https://webvr.info/

And optionally for the current attempt of standard (that will be succeeded by openXR)
http://www.osvr.org/

maxime-coutte, 2018-01-21:
Thanks for the links! I will take a look and work on it !

LifeIsStrange, 2018-01-21:
Good luck ! That will not be easy but the advantages of those specs is that with those standards you benefits from the gpu drivers optimisations that were oculus only before !

As for webVR, you may look at the servo opensource implementation, and probably ask a few questions to the dev mortimergoro :
https://blog.mozvr.com/webvr-servo-architecture-and-latency-optimizations/

But if you don't reimplement yourself the webVR I believe you should port a browser to your headset.
I really see now the need to have an opensource VR OS.. Maybe there already exist VR Linux distributions ? That would save all those issues because Linux already support webVR and OSVR, and will support openXR.
If such a distribution doesn't yet exist that could be your project, and that would be really badass.

Anyway the scope of those goals being big, I heavily suggest you to do a crowdfunding like had done oculus to be known.

maxime-coutte, 2018-01-21:
A crowdfunding without selling anything right ?

maxime-coutte, 2018-01-21:
If you want to get involved, I just created this channel for Relativty: https://discordapp.com/invite/W9VKbjU

closed 2018-01-21 by maxime-coutte

rpavlik, 2018-01-21:
OSVR support is a great idea (I'm the maintainer, so biased, but...).  Haven't looked too much at your device yet so I don't know the details of the hardware, but we designed the plugin system to be very easy to connect to whatever hardware or existing sdk you may or may not have. And, one way or another we're planning to keep all existing device support - OSVR started as the most widely compatible framework and it will stay that way - even once OpenXR is public (and I'm on the committee) so don't worry about that part.

(Oh and btw, a ~$250 headset that supports, e.g. OSVR and steamvr does not, in fact, sell like hotcakes, even before Oculus and Vive price cuts happened, from experience, and is probably not a great source of profit en-masse, just as a caution, but don't let me dampen your enthusiasm for VR or your do-it-yourself spirit :D )

rpavlik, 2018-01-21:
FYI, there's some simple sensor fusion code included in OSVR/VRPN that should be reasonable usable on an MPU6050 (at least, if you have it configured approximately like the Oculus dk/dk2 -they opened those designs after several years so I've seen lots of clones of them and it works pretty well), as well as some modular camera-based positional tracking code as well as the associated firmware if you want something nearly off the shelf, otherwise you're also welcome to use the Kalman filter framework and code that is bundled with OSVR as well. It's very much doable for students to hack on: I came from academia so it's based on both the best work I've seen and the best papers I've read, and I've included citations to the papers in the code.

While OSVR and the commercial clients of Sensics is my day job, and those clients have me pretty busy, I'll try to offer at least a pointer if you get stuck and post a question in issues or something. (Just don't use Reddit, I don't use it anymore because the community was somewhat hostile)

(BTW, you can get the tracking camera from the hdk sold separately at osvrstore.com - it has some firmware issues in that it blends frames together a bit, which made our algorithm which used synchronized blinking a bit of a struggle, but it's still a good IR camera sensor that does 100fps, )