relativty/Relativty #5: Clarification on the LCD Screen

nicolasguzca, 2018-01-21:
Hi! This project looks awesome!
Quick question, saw the LCD Screen you guys posted. Should I buy it with the board ( over 100 USD ) or just the screen?

maxime-coutte, 2018-01-21:
If you just buy the screen it will be impossible to connect it to your computer. The board is a MIPI to HDMI adaptor. I'm sorry price increase heavily and I'm searching a solution to offer Screen for less than $80. You can buy a cheaper LCD on alibaba.com but you will have to deal with factory.

nicolasguzca, 2018-01-21:
Got it, thank you for your answer!

Well the price is very good for a 2K display, it looks very high quality.

maxime-coutte, 2018-01-21:
I just created this channel where you can ask anything anytime, https://discordapp.com/invite/W9VKbjU.

closed 2018-01-21 by maxime-coutte