relativty/Relativty #12: Some questions

Akito13, 2018-01-23:
I was talking extensively with my friend about this project, because we find it very interesting. When we checked out the components needed for the system, we figured that the Arduino Due would be quite an overkill for its job here. Did you choose this specific device on purpose or did you just pick an Arduino Due because it was more convenient in your situation?
According to our results a way weaker micro controller could be used in this scenario.

Secondly, how important is it that the screen has at least 1440p? Did you test it with a lower resolution to see if it is garbage or acceptable?

maxime-coutte, 2018-01-23:
Lower resolutions are acceptable 🙂 We use Arduino Due for the Native USB. 

I would love to talk more about the project with you. Maybe we can chat on the discord Fellowship or you can ping me ? maxime@relativty.com

Akito13, 2018-01-23:
I sent you an e-mail.

closed 2018-01-23 by Akito13

eracoon, 2018-01-23:
An Arduino pro micro would do the trick also I believe then since it's an
HID device with native USB. No?