relativty/Relativty #13: Switch from MPU-6050 6 axis sensor to MPU-9250 9 axis

luckcolors, 2018-01-28:
As previously discussed on discord, using a gyroscope without a fixed reference point causes yaw drift to happen to on the Z axis on long sessions (this naturally happens due to earth's rotation).
Switching to the MPU-9250 a 9 axis sensor gives us access to the (magnetometer) compass wich allows us to get the referece point on earth's north.
Doing this change will make the expense fom the sensor from 0.80-1$ to 2$, (in my opinion something that is worth the extra 1$ :) ) .
We could also use alternative sensors like the BNO055 but it's 10$ on aliexepress.

eracoon, 2018-01-28:
that's a very interesting point.
but won't any stronger magnetic field throw off the sensor a bit?

maxime-coutte, 2018-01-29:
We should do this.
I will order an MPU-9250 today.

luckcolors, 2018-02-05:
Umm, soo what's the decision on this? If you decided that we should switch, we should update the Readme as well and make the necessary code changes.

maxime-coutte, 2018-02-05:
@luckcolors I haven’t received my MPU9250 yet. You can make the necessary code changes, PR,  and I will update the code when tested.

delp, 2018-02-08:
I ordered an MPU9250 as well and will perform testing with it in my build when it arrives.

luckcolors, 2018-02-08:
Awesome! When you get the sonsor in your hands we should start to discuss about the code required to make the sensor and the error correction work.

kasokz, 2018-02-13:
Instead of proposing the MPU-9250 which contains a MPU-6500 as improvement, couldn't we use the MPU-9150? MPU-9150 contains the MPU-6050, which we are already familiar with, and an additional magnetometer. 
The magnetometer of the MPU-9250 is a little better than the one inside the MPU-9150, but that shouldn't make much of a difference.
Source of the statement here: http://www.sureshjoshi.com/embedded/invensense-imus-what-to-know/

luckcolors, 2018-02-15:
I think if we can get a newer sensor the better, if you notice at the end of the MPU-9150 page's there's a note saying it's discontinued.

Ybalrid, 2018-02-17:
The Oculus DK1 used a magnetometer to correct yaw drift (and the included accelerometer to have a vertical reference too). But it really used to be a pain to calibrate correctly...

I think the direction you should go toward, it you want to make this project evolve, would be to add an external optical tracking system, and add positional tracking too.

You would kill two birds in one stone as you would have two sources for the user's head orientation, one being absolute, and not integrated over time...

It could be as simple as a few LEDs on the headset and a cheap webcam watching the user. But it would require more processing on the computer side. If implemented, the hardware cost would still be quite low, and it could be just an "additional" upgrade to the base system.

Just some random thoughts ^^"

ToastTea, 2018-02-17:
Here are a couple open-source projects for head tracking so you don’t have to reinvent the wheel.  http://www.free-track.net/english/
And
http://www.edtracker.org.uk/

delp, 2018-02-17:
@Ybalrid I've been playing around with trying to get positional information from a xbox 360 Kinect. I have it working physically and though it's nice hardware for a good price (Usually about $10 at the thrift store) I have had a really difficult time with all of the software made for it (at least on linux.) None of it has aged very well. I spent several weeks battling horrible dependency problems and never succeeding in getting anything beyond LibFreeNect to work. So I think I've ruled out the Kinect as a road I want to go down, personally.

But I think I agree with you, in my mind the ideal solution for positional tracking would be a simple cheap IR/Camera setup that also sends info to the computer. I would like to try this at some point once the main headset has some more builds and testing.

At this point we are really just in need of as many people as possible to try 1) sourcing affordable parts 2) assembling from the info we have 3) documenting the good and bad and any build/software problems.

xmak, 2018-06-24:
Is there any progress with integrating MPU-9250 or MPU-9150?

AdmiraalHugo, 2019-04-17:
With the change to the STM32 board, a upgrade to the MPU-9250 is planned. We hope to implement this future in a few months.

JustinS-B, 2019-05-31:
We now have headset-ready code available to test out the MPU-9250 and MPU-9255 based boards.

closed 2019-05-31 by JustinS-B

bentalebahmed, 2019-07-12:
you should try Microsoft Kinect , i have tried it for positional tracking and then sending data to android phone that supports VR also added a haptic feedback system, you can check this link (research poster submitted for VRST 2018 in Tokyo ): 
https://hal.archives-ouvertes.fr/hal-02052104/document