relativty/Relativty #15: Replace Arduino Due with ESP32

fharni, 2018-01-30:
Why you don't replace the Arduino Due with an ESP32. It is much faster and has a lot more flash memory. In china you can get a [developer board](https://de.aliexpress.com/item/ESP-32-ESP-32S-Entwicklungsboard-WiFi-Bluetooth-Ultra-niedrigem-Stromverbrauch-Dual-Cores-ESP32-Bord/32844181786.html) for about 5 €.
Here is a [comparision chart of some boards](https://hilo90mhz.com/arduino-esp32-esp8266-101-speed-test-comparison-chart/)

maxime-coutte, 2018-01-30:
Thanks I will take a look at it 😊

fharni, 2018-01-30:
Next advantage is onboard WiFi an Bluetooth.

maxime-coutte, 2018-01-30:
First we will release the SteamVR compatibility with the current Hardware and then maybe switch to ESP32 + MPU9250 😊

maxime-coutte, 2018-01-30:
Thanks a lot for the suggestion, it will help a lot.

Krossom, 2018-02-01:
but how is the development with ESP32? language... etc

fharni, 2018-02-01:
You can program it in the same way like the Arduino. Here is an instruction how to [get started](https://github.com/espressif/arduino-esp32) with the Arduino IDE and the ESP32.

maxime-coutte, 2018-02-05:
@fharni If you want to start working on a ESP version of the headset you should join our Fellowship, some peoples are also working on this. We would love to make the headset compatible with ESP !

closed 2018-02-05 by maxime-coutte