relativty/Relativty #16: Update Headset Form

Lootgeier, 2018-02-03:
The Form of the Headset doesnt look comfortable at all.
Is there a possibility to update it so it sits firmly and is not so boxy?
Maybe compatible with the Foampad from the Samsung Galaxy VR (7€) foam piece.
And where to get the headstraps and stuff?
Maybe it would be possible to merge this project with "WiFiBroadcast" from Raspberry so it gets fully Wireless with a powerbank. Im really looking to the improvements but im not good at 3D modelling but im trying to make some games for it like a "Raft VR" version and "MarioKart 3D" or "Prison Escape" so we can fully enjoy this nice project.

Thanks for your great work and keep having fun in coding :)

maxime-coutte, 2018-02-05:
If you 3D print arduino and screen controller bumpers you can stick them to a cardboard. 

That would be awesome if you create games for the headset! Have you joigned our Fellowship? I’m sure some Fellows could update 3D models :-)

maxime-coutte, 2018-02-08:
Feel free to re open the issue if needed :-)

closed 2018-02-08 by maxime-coutte