relativty/Relativty #19: support for Unreal Engine

xmak, 2018-02-10:
Do you plan to support Unreal Engine?

I would really like to see this happen and I'm willing to participate with my modest knowledge of UE.

maxime-coutte, 2018-02-10:
We are currently working on stramVR compatibility. But if you want to help with UE that would be awesome !

Feel free to join our Fellowship so maybe you will find some people able to help you.

xmak, 2018-02-10:
How do I join your Fellowship? Do you have some group or forum where you exchange info and ideas?

As I said, my knowledge of C++ coding for UE is pretty modest (I have only worked on some experimental C++ code for network communication and for using MS Kinect in UE), so I would like to find someone who is more experienced to join me on this. I could give a fast track introduction in coding for UE to someone who is interested, and do a share of coding.

I'm planing to make your VR headset here at our local library with help from some of my collegues, so when that works out I can also do the testing.

P.S. You guys rock... your VR headset and the whole idea of making it affordable is one of the most interesting DIY things I have seen lately.

delp, 2018-02-10:
The discord chat where we mostly coordinate is linked at the bottom of the main readme but it's sort of hard to see.

Come about and let's see if we can pair you with someone who wants to work on this.
https://discord.gg/W9VKbjU

Ybalrid, 2018-02-17:
I dug a little bit into your FastVR and WRMHL repositories. As far as I can tell, you are just sending orientation data from the headset to the computer via 250000 baud RS232 (Serial) link from your Arduino board?

It should be easy to make this usable in Unreal, or anything else for that matter. 

I agree that a SteamVR integration is probably the ideal route for any new headset on the market (at least until the guys from the Khronos group release [OpenXR](https://www.khronos.org/openxr), since it would make it compatible (at least for getting the images and your current 3 axis tracking) with existing content.

xmak, 2018-03-04:
To close this issue, we have SteamVR support now (via OSVR):

https://github.com/relativty/relativ-core

I haven't tried it but some of the guys did, and it seems to work. I will confirm when I try it with Unreal Engine.

closed 2018-06-06 by al-haruun