relativty/Relativty #21: [feature request]inertial tracking using the onboard imu

jwdinius, 2018-02-12:
After looking through the implementation, I could not find an inertial tracking algorithm for your headset.

Would you like one?  I've found an [algorithm](http://msl.cs.illinois.edu/~lavalle/papers/LavYerKatAnt14.pdf) that I believe I could implement in a few weeks.

delp, 2018-02-12:
Hey, welcome.

We've just started trying to figure out how to organize the project as a community project and collaborate with people who want to contribute. Many of us are still building our headsets (waiting on parts from Shenzhen ;) )

Anything you'd like to work up would be awesome and I'm sure we can find users to test it.

If you have questions or want to try to recruit people to work with, you can check out the Discord channel (Link at bottom of the main Readme.)

Would love to see what you have, go ahead and submit a pull request when you can!

delp, 2018-02-12:
I'm still in the process of trying to come up with an "Issues" template, how to categorize them, and a collaboration guideline, etc.

jwdinius, 2018-03-09:
It looks like the MPU6050 provides all of the raw data that is needed through its api.  I am putting the code together to add this feature right now.

jwdinius, 2018-07-02:
It looks like this has been merged into master.  Ready to close the issue?

closed 2018-09-30 by AdmiraalHugo