relativty/Relativty #36: Serial Output is all zeros!

kaushal1214, 2019-01-18:
The output on the arduino serial is something like-
`0.00 0.00 0.00 1.00`
And it's not changing at all. Plz help
[image]

kaushal1214, 2019-01-21:
Hi,
Yes, I'm using MPU6050 and Arduino Due. I think I have connected it correct, because I tested MPU6050 with an example sketch, which are available in arduino IDE. I selected "MPU6050_Raw" example under the "Example from Custom Libraries" category. 
At the output I can see the values from gyro changing continuously with motion. It means the MPU6050 and its connections are correct, right?
And I have not used SteamVR till now, as I am stuck with non-functioning of MPU6050.
Hope it helps you to understand my problem.

Thanks

AdmiraalHugo, 2019-01-23:
Hi,

Sorry for the late answer. Yes, that means the MPU6050 is connected good, so that is good to know. Can you contact me on hugo@relativty.net? So I can send you a different code, that might work.

Regards,
Hugo
member Relativty Committee

JustinS-B, 2019-01-24:
Also, we have a specific Linux group on the Relativ Discourse, where we can help you more easily

AdmiraalHugo, 2019-01-24:
If you do have a discord account, you can join our group with this link; https://discord.gg/s9YqY4P

closed 2019-04-17 by AdmiraalHugo