relativty/Relativty #37: 'VectorFloat16' was not declared in this scope

ivanbag, 2019-03-16:
Hello, trying to compile the sketch I get the following error:

**************************************

main:133:9: error: 'VectorFloat16' was not declared in this scope

         VectorFloat16 down;

         ^

main:134:28: error: 'down' was not declared in this scope

         mpu.dmpGetGravity(&down, &q);

                            ^

exit status 1
'VectorFloat16' was not declared in this scope
 
**************************************

How can I correct it?

Thanks, regards

MicklesC, 2020-06-23:
I know this is over a year old but did you ever figure out a fix for this? I have the same issue.

lunar711, 2020-07-27:
I got it to compile, but the catch is that only Arduino boards that have an internal USB only work (no FTDI or Atmega16U2 as USB controllers). The code compiles if you use Arduino Leonardo and the likes.... (remember to change the "#define INTERRUPT_PIN 2" the pin number '2' to the right Interupt Pin for your board, SDA and SCL depends on what board you use)

```

// Copyright (c) 2017 Relativty

//=============================================================================
// i2cdevlib and MPU6050 are needed: https://github.com/jrowberg/i2cdevlib
// Big thanks to the amazing Jeff Rowberg <3, go check his repo to learn more MPU6050.
//=============================================================================

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project

#include "Relativ.h"

#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"

#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
#include "Wire.h"
#endif

MPU6050 mpu;
Relativ relativ; 

#define INTERRUPT_PIN 7
// set these to correct for drift - try small values ~0.000001
#define ALPHA 0.f // correction for drift - tilt channel
#define BETA  0.f // correction for drift - yaw channel

// IMU status and control:
bool dmpReady = false;  // true if DMP init was successful
uint8_t mpuIntStatus;   
uint8_t devStatus;      // 0 = success, !0 = error
uint16_t packetSize;    
uint16_t fifoCount;     
uint8_t fifoBuffer[64]; 

Quaternion q;           // [w, x, y, z]

volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high
void dmpDataReady() {
    mpuInterrupt = true;
}

void setup() {

    #if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
        Wire.begin();
        Wire.setClock(400000); // 400kHz I2C clock. Comment this line if having compilation difficulties
    #elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
        Fastwire::setup(400, true);
    #endif

    /**
    * Relativ::startNative() is for 32-bit ARM microconrollers with Native USB like Arduino DUE (recommended) 
    * Relativ::start() is for NON-NATIVE USB microcontrollers, like Arduino MEGA, Arduino UNO, etc. (slower)
    */
    relativ.startNative(); //Compile error? change this to relativ.start() if you have an UNO, MEGA, etc... 

    mpu.initialize();
    pinMode(INTERRUPT_PIN, INPUT);
   
    //If you are getting "SerialUSB was not declared" when compiling, change
    //the following line to Serial.println(.......)
    SerialUSB.println(mpu.testConnection() ? F("MPU6050 connection successful") : F("MPU6050 connection failed"));

    // configure the DMP
[... 80 lines cut ...]
        Quaternion d_yaw; // just default to identity if no magnetometer is available.

        // COMPLEMENTARY FILTER (YAW CORRECTION * TILT CORRECTION * QUATERNION FROM ABOVE)
        double tilt_correct = -ALPHA * 2.f * acos(d_tilt.w);
        double yaw_correct  = -BETA  * 2.f * acos(d_yaw.w);
        //Quaternion tilt_correction( cos(tilt_correct/2.f), -down.normalize() * sin(tilt_correct/2.f), 0.f, down.normalize().x * sin(tilt_correct/2.f) );
        
        Quaternion tilt_correction( cos(tilt_correct/2.f), -down.getNormalized().getMagnitude() * sin(tilt_correct/2.f), 0.f, down.getNormalized().x * sin(tilt_correct/2.f) );
        Quaternion yaw_correction( cos(yaw_correct/2.f), 0.f, sin(yaw_correct/2.f), 0.f);

        // qc = yc * tc * q
        Quaternion qc;  //(tilt_correction.w, tilt_correction.x, tilt_correction.y, tilt_correction.z);
        qc = qc.getProduct(yaw_correction);
        qc = qc.getProduct(tilt_correction);
        qc = qc.getProduct(q);

        // report result
        relativ.updateOrientationNative(qc.x, qc.y, qc.z, qc.w, 4);

        //If your microcontroller is non-native usb, change the above line to:
        //relativ.updateOrientation(q.x, q.y, q.z, q.w, 4); 
    }
}

```

RasPI-AG-WSG, 2021-07-01:
I have the same problem. I'm using an Arduino DUE. I can't find the VectorFloat16 in any of the used libraries. There's only VectorFloat. But the program doesn't work with the VectorFloat function.

lunar711, 2021-07-01:
Sorry but I've given up on this project ages ago.... If I remember this uses OSVR (Which has been dead for a long time now thanks to SteamVR always updating). This headset only tracks rotation only, like anyother mobile VR headset.....