relativty/Relativty #47: Clarification about ATW

off99555, 2019-09-09:
Does this headset support Asynchronous Timewarp (Reprojection)?

AdmiraalHugo, 2019-09-09:
Hello @off99555 ,

If i'm right, this is build into OSVR, the software we use to connect our headset to SteamVR.

Regards,
Hugo
~ Relativty Committee

off99555, 2019-09-10:
Yes. OSVR seems to have timewarp feature. You can check it in your headset screen by rotating it fast and see if there are black borders. If there are black borders then it means there is timewarp.
The black borders will appear on the left side of the screen when you rotate to the left. And the black borders will appear on the right side of the screen when you rotate to the right. How big the black border is depends on the latency.
Does your headset have black border when rotating fast?