relativty/Relativty #48: [FEATERE REQUEST]can't connect relativty.net because China's net

JOKER-404, 2019-09-28:
As the result of living in China, I can't connect relativty.net for more detail to build this great project.
Could you please post the content in that web site on github ?

maxime-coutte, 2019-10-28:
Do you mean relativty.net is blocked in China?

JOKER-404, 2019-10-29:
yes, it is blocked.

AdmiraalHugo, 2019-10-29:
You can use a VPN, or the "Anonymous View", from startpage.com. It will work on that. I don't have a lot of time right now, due to school and work.

JOKER-404, 2019-10-30:
thanks，but these methods don't work in my country because of gov publishes lots of laws and installs firewalls in ISP.

AdmiraalHugo, 2019-10-30:
I can't do a lot about the block, I have mailed a lot with my hosting provider, and he changed the IP like 4 times. I'll see if I can upload the website to github when I have time, like I said in my privious comment.

JOKER-404, 2019-10-30:
Really appreciate it.

weiyongwill, 2019-11-12:
also not work via VPN in Japan