relativty/Relativty #50: [FEATERE REQUEST] Compatible SteamVR games, controllers and laser pointer support

Sgw32, 2020-02-12:
Hi,

Built my own headset! Looks pretty good and works. Thanks a lot for this project. 

Anyway, I am unable to play most of Steam games with this setup:

 - Relativ headset
 - Xbox360-compatible controller

The laser beam just do not appear even in SteamVR Home application. It is also impossible to navigate in a dashboard. What games can you advise for your headset? I've found an interesting space simulator - but it is only game I've found. 
Either game is playable through mouse/keyboard/joystick, either it is connected to SteamVR and it is impossible to play the game. I just see what's aroung in this headset - which is pretty good, but I am interested in making it playable. So, I think, users of the Headset need from you:
1) List of compatible games
2) List of compatible tracker and controller hardware for using with SteamVR and Relativ.

JulianGro, 2020-09-21:
I am pretty sure VRChat works with just an xbox controller. (Windows)
Vircadia seems to work with a keyboard, so I would assume an xbox controller would work as well. Nonetheless Vircadia is an open source project as well, so it might be a good fit.

okawo80085, 2020-09-21:
this was caused by the lack of hmd's bindings, and im guessing by osvr being old af

but now relativty has its own steamvr driver, the hmd's laser pointer should work now(with a controller per say)

if you still find some game not working with it by all means let me know