relativty/Relativty #53: [FEATERE REQUEST] Use correct industry terminology for what this project is developing

ObsidianX, 2020-08-19:
**Is your feature request related to a problem? Please describe.**
This project is using the wrong terminology in regards to the tracking technology being developed.  The sensors used are not "6DoF" or "9DoF" as described.  Using correct terminology will clear up a lot of confusion around the project's current goals and released software.

**Describe the solution you'd like**
"#DoF" stands for "Degrees of Freedom" and is a term used to describe how the user can move about in VR.  The tracking system being developed here only provides *3DoF*: Pitch, Yaw, Roll.

6DoF means that on top of rotation the user is also able to translate around the VR space in X, Y, Z directions.  Therefore there are now 6 degrees of freedom: Pitch, Yaw, Roll, X, Y, Z.  As far as I can figure out there aren't additional degrees in which to unlock to achieve "9DoF".

The sensors you're developing are multi-_axis_ sensors in the pursuit of higher quality "3DoF" tracking.  The 3 sensors utilized in this project provide a 9-_Axis_ tracking solution and only provides more accurate rotational tracking.

**Additional context**
https://creator.oculus.com/learn/vr-glossary/

siddacious, 2020-09-11:
I appreciate the intent to clarify and standardize terminology however IMHO you are mostly right but not completely right. My only nit-pick is  The MPU6050 includes two three-axis sensors and support for a third.

The **accelerometer** measures linear acceleration/translation and the *gyroscope** measures angular velocity. That's 6 degrees of freedom as you defined it. 

I agree that unless they've figured out how to fold spacetime, Invensense and everyone else claiming to have a "9DoF" sensor is letting their marketing team have a bit too much leeway for my taste. Magnetometers are sometimes used to address drift in angular velocity but extrapolating from what you said, that's still only 6-DoF.

With regards to my larger _agreement_, only just now having seen this project I'm going to assume that they're using the 6-DoF IMU/Gyro+Accel to provide 3-DoF tracking, so they should be explicit and clear about that.

capnmidnight, 2020-09-21:
It's "standard" within the IMU market to list the DOF as a sum of all the DOFs of the individual sensing elements. Accelerometer+gyroscope+magnetometer ICs are *always* called 9-DOF.

VR headsets should be labeled according to the degrees the expose to the user. Using a single IMU, you'll only be able to achieve 3-DOF.

capnmidnight, 2020-09-21:
That said, have you considered using the Inte RealSense T265 tracking module for true 6-DOF tracking, and if so, what was your determination?

https://www.intelrealsense.com/tracking-camera-t265/

siddacious, 2020-09-29:
I agree with @capnmidnight that while it might grind my gears slightly, 9-DOF is the commonly used term for such devices. To me this means that if you feel the need to call out the dof-ness of the sensors, 9-DOF is "correct" insofar as it effectively communicates what they are to people that are familiar. If this is worth mentioning is not something I have a stake in. Though it might confuse laypeople.

The tracking solution however is, as I think we all agree, 3-DOF.

ObsidianX, 2020-09-29:
Given the above arguments I think it ultimately depends on the context.  This project is developing a VR headset and the end result is a 3-DoF or 6-DoF product.  The IMUs, cameras, lighthouse sensors, etc, used to achieve that isn't relevant to the project's deliverables and since this is going to hopefully be listed among all other VR headsets should standardize the terminology used to describe the product's features on the rest of the VR industry.

_As a side note I feel given the meaning of a "degree of freedom" it's incorrect for an IMU to use DoF to describe an axis, but that's not relevant to this project ;)_