relativty/Relativty #61: Hardware license

renzenicolai, 2020-09-12:
You've currently chosen to put the GPLv3 license on the project. The GPL is suitable for software, but less so for hardware. In addition please consider how using the GPL essentially complicates any commerical use of your creation. While this indeed keeps your project out of the hands of commercial companies who could run away with the idea it also blocks anyone from creating a proper product based on your work.

What's your end goal for the project? Creating a commercial success for yourself or creating an open VR headset that's made available commercially by several companies?

Anyhow please consider using a permissive license like MIT or BSD for the software and a truely OSHW license like the CERN–OHL-P.

Olm-e, 2020-09-13:
I agree the hardware needs proper OSHW licence

but for the software : GPL does not blocks anyone to create a proper product from here, on the contrary : it would permit a sane redistribution of the code and ensure buyers have all the needed rights on it as companies could not close it down to sell it. It's a good selling point imho. so keep it open and libre ;)

AmmarkoV, 2020-09-15:
GPL3 prevents TiVo-isation ( https://en.wikipedia.org/wiki/Tivoization ) which is a valid concern for a hardware project. If someone wants to create a commercial use its fine as long as he provides what he received as free and open source in the same way.

VinnyCordeiro, 2020-09-15:
GPL is not against commercial use of code/hardware. GPL is against use of code/hardware in a way that prevents the users to study the code and make changes by himself/herself.

If a company wants to make a product with full control of the IP then they should expend their money on R&D and develop their own products, not using open source projects, add some changes, and then lock the users out.

truedat101, 2020-11-26:
Agree 100% with @renzenicolai .   The notion that every business enterprise wants to pillage open source and pull a Linksys/Tivo/MSFT , is wrong.  Imagine an EDU-tech company that is attempting to build platforms for covid era distance learning.  They won't have the expertise to necessarily  build VR HW from scratch, nor the budgets to fund years of effort.   And they may not have the experienced resources to successfully contribute back to open source.  But they might be willing to fund/sponsor development.  

Speaking not from a podium of ideology, but as someone who has worked with companies who attempt to take proprietary their own proprietary work into open source, there are numerous challenges to GPLv2 or v3 that is 99% radioactive for a legitimate business enterprise wanting to build on top of work developed by a 3rd party.  Companies do it and manage to comply, but most of the time this will be in modifying an existing hardware driver or some other aspect of a linux kernel.  Attempts to build a viable product around a GPLv2/v3 licensed hw/sw will usually get shot down by the legal team on the basis that many in-house legal teams just don't understand open source licensing well and don't like the terms of GPLv2/v3.  

There is a secondary challenge around the question of patents, and GPLv2 does not make an explicit patent grant, while GPLv3 does.  That said, GPLv3 does make an explicit grant, however there is nothing stopping someone from making a claim on infringement and it offers no protection to the recipient of the covered sw/hw.  

Can I propose dual license as an option?  The copyright holder(s) always have an option to offer a proprietary license or alternative open source license (assuming all copyright holders are contributors and they all agree to this).  GPL makes the decision to try out/prototype software/hardware very easy, however, companies who have a legal team that blocks exposure to these licenses, will want to get a different license.  And they will usually pay for this right, at which point the copyright holder can attach additional terms which may achieve the same goal in preventing Tivo-isation without the licensee worrying about the implications of copyleft provisions might have on the greater work they intend to create.  Companies will also want to seek a more detailed set of terms around patent grants if applicable.

A dual license/alternative license doesn't have to go into the license file and can be made available upon inquiry.  Personally I prefer an option of GPL or academic license explicitly in the LICENSE file , and this still encourages contributors to then be compliant with both licenses if they wish to provide patches / pull requests.