relativty/Relativty #64: Single display instead of stereoscopic

errorenousReport, 2020-09-14:
I there a way to create this with a large single display instead of dual displays?  I know many people desire stereoscopic, but I would rather have one larger display instead of locking my eyes into a little lens.  The 6dof control and head tracking I feel would be more immersive without a limited stereoscopic display and this open source project would be a good testing ground.

okawo80085, 2020-09-21:
with the current design it treats that "dual" display as a single display, windows sees it as one, its separated into 2 panels only physically, the software sees it as 1 whole display. also current software setup *can only* work with a single display, as in steamvr's vr view will be only visible on 1 display

and about
> The 6dof control and head tracking I feel would be more immersive without a limited stereoscopic display and this open source project would be a good testing ground.

what might you mean by that?

bc either way you will get a stereoscopic image(2 eyes, 2 different camera views), for vr you don't get to choose

also how is this setup limiting??? if htc, valve, oculus and others use 2 panel displays maybe there is a good reason for that

capnmidnight, 2020-09-21:
I think OP is talking about having a single, large view, like a monitor attached to your head, with no stereoscopic rendering, not just the number of display modules inside the headset.

No VR headset on the market does this for a very good reason: focal distance.

If you try to make a lens system that focuses both eyes individually onto a single, small display panel, you end up with *huge* amounts of lens distortion, such that you'll never get both eyes lined up on the display panel correctly. You'll have all of the pixels of the display going into each eye, but the relative position of each pixel will be massively out of place and you'll just end up with double vision.

If you forgo the lenses completely, you have to push the display out far away from the face. This increases the angular momentum of the display, meaning whipping your head around gets a lot harder and a lot more tiring. Also, just holding your head up gets a lot more tiring, with or without a counterbalance. Plus, to get a full field of view, you'd have to use a much larger display, which exacerbates all of those weight issues.

No, you can't do a single, monoscopic view headset with the hopes of achieving wider FOV. You'll always end up needing some kind of lens system, which will always split the view between different halves of the display module (or two separate display modules). At which point, you might as well make it stereoscopic.

okawo80085, 2020-09-21:
and none of the frame delivery components available for drivers support monoscopic views

marcdraco, 2021-05-03:
As a movie guy (games are a secondary consideration for me, an important one though) I can see what the OP's getting at but I think they misunderstand how binocular vision works.

@errorenousReport We need lenses on a headset for essentially the same reason as we need them on, say a microscope. Our eyes just can't focus on objects a few mm in front of them without creating serious eye strain. Try looking at something and moving it closer and closer to your head and you'll see that as things get closer it's difficult and eventually impossible to focus on it.

Binoculars and dissection microscopes have two eyepieces for the same reason. The (fresnel) lenses do in our current HMDs - they move the focal point far enough into the distance (as our eyes see it) so they are able to focus without  eye strain. It's doable with glass lenses but they cost a lot more and even with the best high-refractive index optical glass we have today, they will be considerably thicker and much heavier too. 

@capnmidnight sums it up well but just to add to that it's important to realise that our brains don't see the world as a flat plane either. We see a flat plane because it's, well, flat. Our brain figures that out when it creates the perception of what we're looking at (similarly it also focuses across our entire field of vision but makes it *appear* to us that everything is in focus). In fact, our world (to our eyes) is upside down, but our brains flip it the right way up. There's a lot of neuroscience around this that I won't delve into here but it's damn clever stuff.

The Nintento 3Ds managed something using a single display with (I think, a fresnel) with fairly poor results - but that was for kids who were more interested in the novelty. Perhaps this is what you were thinking about? Or maybe you're thinking along the lines of HSBS that's used in el-cheapo headsets like Google cardboard (more of that in a moment).

There is a _potential_ solution to all of this (in research/academia) but there's no consumer version that I'm aware of as yet and may never be one. As I recall it involves using lasers to "fire" light rays directly into our eyes. Unlike the "high-power" lasers that can damage the eye, these use visible light at a very low power and are directed so they enter the eye and hit the retina already in focus. Rather like cutting out the middle man. It sounds like science fiction and I won't pretend that I understand it myself - although I suspect the military already uses something similar for HMDs.

For now, we're pretty much stuck with the available technology - because if something better appears, it will likely be subject to patents, crippling license fees and only be practical for the big guys. In Open Source, we have to innovate but optics and optical electronics are (again) something that only companies with bottomless pockets are able to dabble in.

What I would like to see (and Open Source can innovate here a little) is moving some of the weight elsewhere on the HMD. I think Panasonic demonstrated something recently but that's still in development. In essence that appeared to use two circular TFTs (square would work just as well) with lenses over each eye as in current designs but more of a John Lennon look. It appeared from the glance I got that all the electronics other than the displays had been moved away from the face - which is really the problem most of us have with HMDs that the b*ggers are so darn heavy.

I have one of those nasty HMDs that you drop a phone into - Google cardboard sort of thing and as is, it's quite lightweight despite being sloppily designed. Put a phone in it (single display) and it's unusable after a few minutes because it falls off my face.

I don't think Relativity has this issue because it's tethered (correct me please if I'm wrong) but if the team is going to look at something wireless like the Oculus Quest/Go/Quest 2 (etc.) then that battery is going to be a massive problem. I guess this is a discussion for elsewhere though.

Stay safe everyone!