relativty/Relativty #71: About the driver

Jianglf19, 2020-11-12:
Turn on steamvr, no matter whether my gyroscope is moving or not, the screen will jitter violently. I thought the gyroscope was broken, but the data on the serial port monitoring tool is normal
From Google Translate

okawo80085, 2020-11-12:
what sensor are you using?

Jianglf19, 2020-11-14:
Is gy9250(mpu9250 chip)+arduino due

Jianglf19, 2020-11-14:
> 您正在使用什么传感器？
> 
> 您正在使用什么arduino固件？

Is gy_9250(mpu9250 chip) and arduino due

okawo80085, 2020-11-14:
im guessing you used arduino code from `Relativty/Relativty_Firmware/MP9250-HID/MP9250-HID.ino`

if you're using that arduino code, you *have* to change this to `false` in the driver config
```json
...
"hmdIMUdmpPackets":  false,
...
```

Jianglf19, 2020-11-14:
> 我猜你使用了arduino代码 `Relativty/Relativty_Firmware/MP9250-HID/MP9250-HID.ino`
> 
> 如果您使用的是arduino代码，则_必须_`false`在驱动程序配置中将其更改为
> 
> ```json
> ...
>  “ hmdIMUdmpPackets ”：   false，
> ...
> ```

Thank you for your help, it can be used normally

closed 2020-11-16 by thelostgarden